abelectronicsuk/ABElectronics_Python_Libraries #4: Duplicate readings of channels

dhhagan, 2015-02-17:
It appears that when I read channels 1 and 2, I get the same reading. Same goes with all pairs (3,4: 5,6: 7,8). Have you been able to test this with some known voltage? Say a pot or something? I'm just wondering if I'm crazy or not.

abelectronicsuk, 2015-02-17:
Which board and library are you using?

dhhagan, 2015-02-19:
I am using the Delta Sigma v2 board with the DeltaSigma library. Attempting to read all 8 channels..I am currently trying to work through the library and figure out what could be going on. I assume reading channel 1 means reading the differential from pins 1,9 which should be channel1+ and channel1- according the the MCP3424 docs?

abelectronicsuk, 2015-02-20:
You are correct regarding the input channels. We tested the code today with different input voltages on all the delta sigma pi inputs and it is showing the correct different voltages on each one. Please can you post your code so we can try it on a board?

dhhagan, 2015-02-20:
After looking at it some more and performing more tests, it appears one of the ADCs on the board is fried, and it is not a software error at all. I double-checked with a couple more Delta Sigma v2 boards, and this seems to be the case. Sorry for the confusion.

closed 2016-01-23 by abelectronicsuk