abelectronicsuk/ABElectronics_Python_Libraries #10: ADCPi Plus -- can't get example to work

PeterPablo, 2017-07-27:
I am running `Linux raspberrypi 4.9.39-v7+ #1 SMP Wed Jul 26 05:31:05 CEST 2017 armv7l GNU/Linux` and i2cdetect gives the following output:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- 6d -- --
70: -- -- -- -- -- -- -- --
```

Unfortunately, the [example (copied to another file)](https://github.com/abelectronicsuk/ABElectronics_Python_Libraries/blob/master/ADCPi/demos/demo_readvoltage.py) fails with:
```
Traceback (most recent call last):
  File "test_ADC_I2C_MCP3428.py", line 60, in <module>
    main()
  File "test_ADC_I2C_MCP3428.py", line 39, in main
    adc = ADCPi(0x68, 0x6d, 12)
  File "/usr/local/lib/python3.4/dist-packages/ADCPi/ADCPi.py", line 109, in __init__
    self.set_bit_rate(rate)
  File "/usr/local/lib/python3.4/dist-packages/ADCPi/ADCPi.py", line 261, in set_bit_rate
    self.__bus.write_byte(self.__adc1[0], self.__adc1[1])
  File "/usr/local/lib/python3.4/dist-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/usr/local/lib/python3.4/dist-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
OSError: 5
```

As you see, I adapted the address of the second ADC to `0x6d`, as for some (to me currently unknown) reason `i2cdetect` would not detect it reliably on it's default address.

abelectronicsuk, 2017-07-27:
The error you are getting suggests that the library can not access the ADC on the I2C bus.  It may be worth checking the solder joints on the I2C address header for any bridges or dry joints and make sure there is a good connection between the jumpers and the header.

PeterPablo, 2017-07-28:
Thank you for getting back to me so quickly. It turns out I simply had confused the parallel row of the input channels and had swapped GND with Umeas. (GND being the solder points close to the PCB edge)

Now the example seems to run just fine. Thanks again!

closed 2017-07-28 by PeterPablo