abelectronicsuk/ABElectronics_Python_Libraries #15: Error running demo_pwm.py

MagicMods, 2018-06-02:
Hi there
getting this error running the demo
with python 2.7 and 3

Traceback (most recent call last):
  File "demo_pwm.py", line 49, in <module>
    main()
  File "demo_pwm.py", line 39, in main
    pwm.set_pwm_freq(1000)
  File "/usr/local/lib/python2.7/dist-packages/ServoPi/ServoPi.py", line 112, in set_pwm_freq
    newmode = (oldmode & 0x7F) | 0x10
TypeError: unsupported operand type(s) for &: 'exceptions.IOError' and 'int'

Any idea how to resolve this issue?
Thanks in advance

abelectronicsuk, 2018-06-02:
Hi

It looks like the code is having trouble talking to the Servo Pi as it is returning an IOError exception.

Can you check that the I2C bus on your Pi is configured correctly?  We have a tutorial for this at [https://www.abelectronics.co.uk/kb/article/1/i2c--smbus-and-raspbian-linux](https://www.abelectronics.co.uk/kb/article/1/i2c--smbus-and-raspbian-linux)

If you run the command "**sudo i2cdetect -y 1**" you should see the servo pi appear on the list, normally on address 0x40.

Also if you are powering the Servo Pi from the Raspberry Pi make sure you have bridged the PWR Link pads on the board with a solder blob, otherwise it will not have any power and will not be visible to the Raspberry Pi.

MagicMods, 2018-06-02:
Thanks for the swift reply : )
I was using a adafruit pca9685 before so i am confident my i2c config is ok.
I did not however think of soldering the PWR pads..
Cannot solder the pad right away, will report back.

Thanks for your support

MagicMods, 2018-06-02:
Now working as expected : )

Thanks for the swift support!

closed 2018-06-02 by MagicMods