abelectronicsuk/ABElectronics_Python_Libraries #18: Timeouterror causing values to freeze

Felhamed, 2019-03-11:
Hello,

I am using the ADC Differential Pi to do some telemetry on batteries. I have a thermistor/current sensor and voltage sensor (all calculated by reading voltage drop). However, when I get a timeouterror I see that the values freeze on the last one before the timeout. Is there something I can do to avoid this?

```python
    def __read_adc(self):
        self.__adc = ADCDifferentialPi(0x6a, 0x69, 18)
        self.__adc.set_pga(1)
        self.__fail_count = 0
        while not self.stop:
            try:
                timestamp = datetime.now()
                self.dataset[0] = ((self.__adc.read_voltage(1) + self.config.vcor) * self.config.Gv)
                self.dataset[1] = (self.__adc.read_voltage(4) - self.config.icor) / self.config.vgain
                if -15 <= self.dataset[1] <= 15:
                    self.dataset[1] = 0
                tcor = self.calc_temperature(self.calc_resistance(self.__adc.read_voltage(3)))
                self.dataset[2] = (self.__adc.read_voltage(8)/self.config.celcius) + tcor
            except KeyboardInterrupt:
                LOGGER.info("Stopping")
                break
            except:
                LOGGER.exception("Unexpected: ")
                sys.exit()
                self.__fail_count += 1
            time.sleep(0.2)
```
```bash
Unexpected: 
Traceback (most recent call last):
  File "/home/pi/scmeter/strom_diagnostic.py", line 82, in __read_adc
    tcor = self.calc_temperature(self.calc_resistance(self.__adc.read_voltage(3)))
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 136, in read_voltage
    raw = self.read_raw(channel)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 199, in read_raw
    raise TimeoutError(msg)
TimeoutError: read_raw: channel 3 conversion timed out
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 50, 769936)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 50, 769936)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 51, 776438)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 52, 782307)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 53, 788786)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 54, 794547)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 55, 800444)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 56, 806269)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 57, 813006)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 58, 815608)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 34, 59, 817479)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 35, 0, 821270)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
    'table': 'testing',
    'timestamp': datetime.datetime(2019, 3, 11, 17, 35, 1, 825087)}
{   'data': [   {   'Current': -217.7699303136222,
                    'Temperature': 14.410154208169065,
                    'Voltage': 47.31267686170213}],
```

abelectronicsuk, 2019-03-12:
Hello

The timeout error in the ADC Differential Pi library normally occurs because the ADC did not return a sample before the timeout timer reached its end.  This can happen for a few reasons such as other processes holding the I2C bus open or another part of your python program is causing a delay.

The easiest way to get around the timeout problem is to increase the amount of the time the  ADC Differential Pi library waits for a response from the ADC.  

In the [ADCDifferentialPi.py](https://github.com/abelectronicsuk/ABElectronics_Python_Libraries/blob/master/ADCDifferentialPi/ADCDifferentialPi.py) file on lines 171 to 180 is the code that controls the timeout period.

 `# determine a reasonable amount of time to wait for a conversion
        if self.__bitrate == 18:
            seconds_per_sample = 0.26666
        elif self.__bitrate == 16:
            seconds_per_sample = 0.06666
        elif self.__bitrate == 14:
            seconds_per_sample = 0.01666
        elif self.__bitrate == 12:
            seconds_per_sample = 0.00416
        timeout_time = time.time() + (10 * seconds_per_sample)`

Change the 10 in the last time to a larger number such as 20 or 50 and this should solve the problem.

`timeout_time = time.time() + (50 * seconds_per_sample)`

Felhamed, 2019-03-13:
Hi, 

Thanks for that info. I have increased to 20 then 50 and it seems to be more reliable but it still happens and causes some issues. Maybe I have to handle the timeout better. I notice the library is written in python2.7 and I am using 3.5, could switching to 2.7 possibly resolve the issue?

Thank you

abelectronicsuk, 2019-03-13:
The library should work on python 2.7 or 3.5.  Do you have anything other devices connected to the I2C port that could be holding the port open?

Felhamed, 2019-03-13:
I do not however I just noticed another error:

```
Unexpected: 
Traceback (most recent call last):
  File "/home/pi/scmeter/strom_diagnostic.py", line 86, in __read_adc
    self.dataset[1] = (self.__adc.read_voltage(4) - self.config.icor) / self.config.vgain
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 136, in read_voltage
    raw = self.read_raw(channel)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 184, in read_raw
    __adcreading = self.__bus.read_i2c_block_data(address, config, 4)
OSError: [Errno 5] Input/output error
```
I'm trying to watch for when the timeout occurs to restart the thread, that may fix it.
Thanks.

abelectronicsuk, 2019-03-13:
An Input/output error normally means the library is having trouble communicating with the ADC over the I2C bus.

Try checking all of the solder connections on the ADC Differential Pi.  A dry solder joint or bad connection could cause an I/O error.

You could also try running the command "sudo i2cdetect -y 1" to check that both ADC chips are appearing on the correct I2C addresses.

Felhamed, 2019-03-14:
I will try with a backup ADC board I purchased to see if the i/o errors go away but currently I get a couple of these errors:

```
Unexpected: 
Traceback (most recent call last):
  File "/home/pi/scmeter/strom_diagnostic.py", line 88, in __read_adc
    self.dataset[1] = (self.__adc.read_voltage(4) - self.config.icor) / self.config.vgain
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 136, in read_voltage
    raw = self.read_raw(channel)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 184, in read_raw
    __adcreading = self.__bus.read_i2c_block_data(address, config, 4)
OSError: [Errno 5] Input/output error
Unexpected: 
Traceback (most recent call last):
  File "/home/pi/scmeter/strom_diagnostic.py", line 87, in __read_adc
    self.dataset[0] = ((self.__adc.read_voltage(1) + self.config.vcor) * self.config.Gv)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 136, in read_voltage
    raw = self.read_raw(channel)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 184, in read_raw
    __adcreading = self.__bus.read_i2c_block_data(address, config, 4)
TimeoutError: [Errno 110] Connection timed out
Unexpected: 
Traceback (most recent call last):
  File "/home/pi/scmeter/strom_diagnostic.py", line 87, in __read_adc
    self.dataset[0] = ((self.__adc.read_voltage(1) + self.config.vcor) * self.config.Gv)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 136, in read_voltage
    raw = self.read_raw(channel)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 184, in read_raw
    __adcreading = self.__bus.read_i2c_block_data(address, config, 4)
OSError: [Errno 5] Input/output error
```

I changed the timeout to 50 * seconds_per_sample and increased the time.sleep delay to 0.4. I rarely get timeouts now. Also I replaced the sys.exit to restart the thread and it seems to be smooth now. The above errors are just ignored and the ADC just retries and reads voltages successfully so I think its fine now.

Felhamed, 2019-03-27:
I'm having a strange issue where sometimes one of the 2 adc addresses doesn't show up. I get this error:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/scmeter/strom_diagnostic.py", line 81, in __read_adc
    self.__adc = ADCDifferentialPi(0x6a, 0x69, 18)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 130, in __init__
    self.set_bit_rate(rate)
  File "/usr/local/lib/python3.5/dist-packages/ADCDifferentialPi/ADCDifferentialPi.py", line 298, in set_bit_rate
    self.__bus.write_byte(self.__adc1_address, self.__adc1_conf)
OSError: [Errno 121] Remote I/O error
```

 sudo i2cdetect -y 1 shows
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- 69 -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    

I should be seeing 6a as well. Do you think this is solder damage?

abelectronicsuk, 2019-03-27:
If the problem only occurs some of the time then that does sound like a bad solder connection somewhere on the board.  Try reflowing all of the solder joints on the address select header and the GPIO header.  You could also reflow the pins on the ADC chip in case that has a dry joint.

closed 2020-02-02 by abelectronicsuk