abelectronicsuk/ABElectronics_Python_Libraries #19: MPC23017 works Thinkpad Laptop i2c bus but it seems like it times out when writing to it too fast

jardiamj, 2019-04-07:
First of all, thanks for all the information and code you provide for the IOPi plus board, it makes it working with it so much easier and enjoyable.
I know the IOPI Plus board is not intended to be used this way, but I am using my Thinkpad laptop VGA i2c bus to interface with it. The python library works quite well, but I had to place a very short time.sleep(0.01) after __bus.write_byte_data() and other bus interactions because if I don't it seems like the board can't catch up and SMBus() throws me an IOError: [Errno 6] No such device or address.

Is there a more elegant way to deal with this?

Jardi.

P.S.
My reason for trying this is that I am planning on moving my Home Assistant instance from the Pi to an Intel NUC but I have a whole bunch of doors and windows sensors interfacing with it through an IOPi Plus board which I would also like to adapt to the NUC. My plan is to convert your IOPi.py library into a more generic MPC23017 library that can be use in any other linux system with an I2C bus, and since I don't have a NUC yet I am doing some testing using my Thinkpad's VGA port.

abelectronicsuk, 2019-04-07:
Hello Jardi

According to the [datasheet for the MCP23017](http://ww1.microchip.com/downloads/en/devicedoc/20001952c.pdf) on page 7 there are several minimum time delays for the data inputs and stop conditions that are needed for the chip to function correctly.

I expect what is happening is where the Thinkpad is running much faster than a Raspberry Pi it is not leaving a long enough delay after the I2C stop condition before sending the next packet.

You could try increasing the I2C bus speed to see if that fixes the problem.  On the Raspberry Pi the IO Pi Plus can run up to 1MHz but depending on the capacitance of your Thinkpad's I2C bus you may find that you can not run that fast.

If increasing the bus speed does not work then adding the time.sleep() after each bus write is probably the only option to make it work.

jardiamj, 2019-04-08:
Thanks for the reply! I figured that was the case.
Also, if you don't mind I'm planning on taking your module, add support for mcp23008 chip and upload it to PyPI with a more generic name.
I don't see any explicit licensing anywhere, so I thought I would ask here if that's ok.
Jardi.

closed 2019-05-07 by jardiamj