abelectronicsuk/ABElectronics_Python_Libraries #20: SPI Bus too fast

christophe-dupriez, 2020-02-02:
**Measured Voltages about 8% too low**
I was measuring 2.46V at ADC input and it was reporting 1.31V. And I wanted the exact value! I slowed down the bus and around 200kHz, the values are correct.

**Recommendation**
Please slow down SPI by default as precision is very important. For speed, you may explain in a comment that it can be augmented but sub-estimate begins to be over 5% somewhere around 1MHz.

** Line of source code **
self.__spiADC.max_speed_hz = (200000)

abelectronicsuk, 2020-02-02:
I have updated the SPI speed to 200KHz on the Expander Pi ADC class so you should see an improvement in ADC accuracy now.

christophe-dupriez, 2020-02-02:
Thanks a lot Andrew! I hope it may help others...

closed 2020-02-12 by abelectronicsuk