abelectronicsuk/ABElectronics_Python_Libraries #21: TimeoutError not defined in Python 2.7

Vince-vd, 2020-05-06:
**global name 'TimeoutError' is not defined**
When the TimeoutError is raised in Python 2.7  it results in a NameError. From what I can tell TimeoutError wasn't introduced until Python 3.3

**Recommendation**
Please define a custom "TimeoutError" class in the beginning of the library for systems using Python 2. It would be very helpful to be able to handle this error in my code where I import the library when this happens. Right now I have to handle it as a "NameError" instead.

abelectronicsuk, 2020-05-06:
Thank you for your message.  I have added a custom TimeoutError class to the ADC Pi and ADC Differential Pi libraries.

Vince-vd, 2020-05-11:
Works like a charm.
Thanks for the quick response and modification!

closed 2020-05-11 by Vince-vd