abelectronicsuk/ABElectronics_Python_Libraries #24: IOPi.py more "generic" constructor

SCiunczyk, 2020-12-21:
I'm playing with IO Pi Zero module connected to NanoPi Duo2 (Armbian Focal) , I can have it working as expected but it need little tweaking. 

The Constructor of class IOPi in line 134
`device = platform.uname()[1]`
is getting the "name" of board but in my case the hostname is returned. And that is obviously useless :-(

My workaround was just to change line 132 to: 
`i2c__bus = 0`
as my NanoPi Duo2 bus is 0.

What do you thing about idea of providing the "bus number" as additional parameter to the constructor. This will easily expand the coverage of your library to many boards without touching other part of library. Because now every board should be somehow "hardcoded" inside the __get_smbus() method.

for example:

`    def __init__(self, address, initialise=True, bus=-1)`

then if bus != -1 don't check the "board type" - just use passed value as i2c_bus.

abelectronicsuk, 2020-12-21:
I have added an optional bus parameter into the IOPi class so you should be able to set the number for the I2C bus when you create an IOPi object.

`iobus = IOPi(0x20, True, 0)`

The above code should create an IOPi object on address 0x20 with initialisation set to true and the I2C bus 0.

Please can you let me know if this works correctly with your NanoPi Duo2?  If it does I will add the same code into the classes for our other boards.

SCiunczyk, 2020-12-22:
Hi Andrew,

thanks for your ultra fast reaction. The IOPi.py with your correction is working like a charm. You can close the issue and add the code to other boards.

I wish you and all ABelectronics crew a Healthy and Merry Christmas and a Happy New Year!

Slawek

closed 2020-12-22 by abelectronicsuk