abelectronicsuk/ABElectronics_Python_Libraries #26: Resistance on ADC Pi

vuh95, 2022-04-20:
Hello. 

First I would like to thank you for your wonderful products.

In the past I have found on the ADC Pi's page about the information of the resistors installed on it (in order to be able to measure voltages >2v). But I forgot to save.

Today when I used ADC Pi to measure the output signal of a pressure sensor (PSE563, with 1-5v output, https://docs.rs-online.com/3ae2/0900766b80df5651.pdf) I didn't get the right result. I discovered this sensor uses a 1000-ohm resistor as the output impedance so it falsified the measurement results.

[image]

I couldn't find the contact section on your website so I wrote this post. I hope you can tell me about the resistor information on your ADC Pi.

Thank you.

abelectronicsuk, 2022-04-20:
Hello

The ADC Pi uses a 10K and 6.8K voltage divider on the input so if your sensor has a 1k impedance that would change the voltage divider to be 11K and 6.8k.

If you multiply the voltage reading from the ADC Pi by 1.0595 that should give you the correct voltage reading.

vuh95, 2022-04-20:
Thank you so much.

closed 2022-04-20 by vuh95