abelectronicsuk/ABElectronics_Python_Libraries #28: Can not install libraries as GitHub removed password support

epiltonRO, 2024-03-01:
Library: [ABElectronics_Python_Libraries](https://github.com/abelectronicsuk/ABElectronics_Python_Libraries/tree/master)/[RTCPi](https://github.com/abelectronicsuk/ABElectronics_Python_Libraries/tree/master/RTCPi)

CMD as per readme
git clone https://github.com/abelectronicsuk/ABElectronics_Python_Libraries.git

Error Msg: Support for password authenication was removed on August 13, 2021
                 See https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloing-with-https-urls
                 FATAL: Authentication failed

abelectronicsuk, 2024-03-01:
Git should not ask for authentication when using the clone command. I tried running the following command in the terminal on a Raspberry Pi and Mac OS and it downloaded the repository without asking for any authentication.

`git clone https://github.com/abelectronicsuk/ABElectronics_Python_Libraries.git`

Are you using git directly from a command line or through another application?

epiltonRO, 2024-03-01:
Thank you for your quick reply. 

The issue seems to have resolved itself, as I have been able to successful download the library using a different WiFi Network.

For completeness, I was using:
Platform:  Raspberry PI Zero W
OS:  Raspberry Pi OS (Legacy) : A stable legacy version of Raspberry Pi OS Bullseye 
        (downloaded from https://www.raspberrypi.com/software/operating-systems/ )
        Installed using the Raspberey Pi Imager (https://www.raspberrypi.com/software/)
Program:  Terminal (included with the OS)  and command line (from thePZW itself).
WiFi Network: Mobile HotSpot via an Andriod Phone (Samsung Ultra 23)

closed 2024-03-01 by epiltonRO