abelectronicsuk/ABElectronics_Python_Libraries #29: Errno16

epiltonRO, 2024-03-02:
From: Elliott
Board: Raspberry Pi Zero V1.1
OS: Raspbian Bullseye
Python Verison: 3.12.2  https://www.python.org/downloads/
RTC Board: https://www.abelectronics.co.uk/p/70/rtc-pi
RTCLib: As per https://github.com/abelectronicsuk/ABElectronics_Python_Libraries/
Installed: As per Tutorial https://www.abelectronics.co.uk/kb/article/30/rtc-pi-setup-on-raspberry-pi-os
               and https://www.abelectronics.co.uk/kb/article/1/i2c-part-2-enabling-i2c-on-the-raspberry-pi
No other boards added & no other GPIO pins used
IDE: Thonny

Tutortial has worked the RTC is keeping correct date and time.  Have confirmed by turning off WiFi and rebooting several times  with WiFi off.

Error when running demo_rtcgetdate.py from RTCPi see attached file Errno16
[image]
The UU at 0x68 indicates that there is another device or driver at the same address.

Debugging Output see attachment
[image]

Steps Tried
Pins soldered and secure
Redid tutorial

Internet search has shown 
https://www.linuxquestions.org/questions/linux-embedded-and-single-board-computer-78/what-is-error-ioerror-%5Berrno-16%5D-device-or-resource-busy-4175599788/
Implies
2 Solutions
1) Remove  driver however I do not know how to deteremine this and how to remove
2) The last post yes it is i2c slave address and to view te i2cdump output

i2cdump output
[image]

From the dump the issues seems to be with i2c-1 

Resolution sort
1) How to remove what is on i2c-1
2) Other causes/ solutions to fix

Thank you for your time and attention

abelectronicsuk, 2024-03-02:
Hello Elliott

Unfortunately due to the way the RTC driver works in Linux when the DS1307 module is enabled, using the RTC Pi as a hardware clock blocks access to any other programs trying to access the DS1307 RTC chip through the I2C bus. This is indicated when you use i2cdetect as the device shows as UU instead of 68. You can use the RTC Pi as a hardware clock or access it directly through the I2C bus but you can not do both at the same time.

To use the Python library with the RTC you will need to disable the hardware clock by reversing the steps you did in our [RTC Pi setup tutorial](https://www.abelectronics.co.uk/kb/article/30/rtc-pi-setup-on-raspberry-pi-os). 

You can do this by editing "/boot/config.txt" and removing the line "dtoverlay=i2c-rtc,ds1307". 

You also need to edit /etc/modules and remove "rtc-ds1307".

In "/lib/udev/hwclock-set" remove the comments around the following lines:

if [ -e /run/systemd/system ] ; then
exit 0
fi

Reboot your Raspberry Pi and the RTC should appear on address 0x68.

epiltonRO, 2024-03-02:
Hello,
Again thank you for your amazingly quick response and clear instructions.
I thought it might have been something like that given it was the only additional board.
Will try and reply.

Thank you once again.

epiltonRO, 2024-03-02:
Hello
I have followed the instructions (or reverse) and confirmed that the output of  i2cdectect is no connection, all address are -- 
Additionally, when running the demo_rtcgetdate.py I receive [Errno 121 Remote I/O error ], which is expected given the no detection from i2cdectect
As an aside:
" You can use the RTC Pi as a hardware clock or access it directly through the I2C bus but you can not do both at the same time."
I thought this RTC was suitable for data logging, that is depolyed without a WiFi connection and the "Date/Time" from the RTC could be used to start/ stop events and to be used as a timestamp in filenames.  
Is this not the case ?

abelectronicsuk, 2024-03-02:
The RTC should be showing up on 0x68 if the hardware clock modules are disabled so I am not sure why it is not appearing on the I2C bus. Try disconnecting the RTC Pi and removing the battery to reset the module and power down the Raspberry Pi. Hopefully when you reconnect it and turn the Pi back on it will appear again on the I2C bus.

You can use the RTC to get the current date and time when the Raspberry Pi is disconnected from the internet. To use it for starting and stopping events you will need a program that regularly checks the date and time and runs events at scheduled times. 

There are two ways to use the RTC Pi.

You can use it as a hardware clock for Linux which automatically keeps the system time accurate when there is no internet connection or after the Pi has been powered down. When using it this way you would use the [datetime library in Python](https://www.w3schools.com/python/python_datetime.asp) to work with dates. The datetime library uses the Linux system time so the RTC Pi would be working in the background within Linux and you wouldn't interact with it directly.

The second option is to use the RTC Pi independently of the Linux with our RTC Python library. This allows you to use the functions in the library to read the date directly from the DS1307 RTC chip. With the RTC library you can access some features that are not available through the Linux hardware clock, mainly access to the 56 bytes of battery-backed RAM that are available in the DS1307. The downside of using it this way is Linux will not have access to the DS1307 so if you are using the Raspberry Pi without internet access it will not have the correct system date when the Raspberry Pi is rebooted.

If you want to use the Raspberry Pi for data logging the best option would be to use the RTC Pi as a hardware clock for Linux by running through the steps again in our [RTC tutorial](https://www.abelectronics.co.uk/kb/article/30/rtc-pi-setup-on-raspberry-pi-os). This way you can use the RTC Pi to keep the system date and time accurate without a WiFi connection and use the datetime Python library or schedule a CRON job to log your data at set intervals.

epiltonRO, 2024-03-03:
Thank you for your reply,
I  have been able to set the i2c (software) mode and tested with i2cietect and demo_rtcgetdate.py which has work.
By  editing "/boot/config.txt" and removing the line "dtoverlay=i2c-rtc,ds1307".

I have been able to the Hardware/ system Clock and tested with python's  datetime library.
By following the RTCPi tutorial.

Thank you once again for your help.

closed 2024-03-03 by epiltonRO