abelectronicsuk/ABElectronics_Python_Libraries #30: Wrong reading on any input

VK1IAN, 2024-12-13:
I have a RPI 2 model B running buster (debian 10) with a ADC Pi V2.2 attached.
I had no end of trouble getting the ADCPi module to work and spent a frustrating few hours bouncing between my RPi and google (which I'm happy to relate if someone is prepared to listen).
Upshot is I ended up using python3 (using python2 threw a "AttributeError: 'module' object has no attribute 'monotonic'" error).
This is the code I am using:
`#!/usr/bin/env python3

from ADCPi import ADCPi
import time

def main():

adc = ADCPi(0x68, 0x69, 16)

    while (True):
        value=adc.read_voltage(1)
        print(value) # print the value
        time.sleep(1)  # Wait 1sec

if __name__ == "__main__":
    main()
`
I've left out a lot of my comments in the above code.
If I connect the analog input to GND, I see 0V
If I connect 5V to the junction of the resistors (applying 5V directly to the analog input), I see 5.06V (near enough to 5V).
If I connect 5V to the PCB input, I see ~4.9V. I've measured the voltage at the output of the divider and see ~2V at the input to the MCP3424. I even tried the input(s) to the second IC and get the same results.
I'm at my wits end trying figure out what is going on.
I know the ADC Pi I'm using is somewhat dated, but I'm keen to use them (I've got a couple) for simple tasks rather than dispose of them.
 Any guidance you can provide would be appreciated.

abelectronicsuk, 2024-12-13:
I am sorry you are having problems with your ADC Pi.

The MCP3424 is designed to support a maximum input voltage of 2.048 volts which is why we added the voltage dividers on the inputs. When you connect your 5V source directly to the ADC input, or any voltage above 2.048V, it will be displayed as the maximum voltage the ADC Pi can support on its inputs which is 5.06V. Connecting a voltage above 2.048V directly to the ADC could damage the chip and reduce its accuracy, or stop it from working, so I would recommend against doing that in the future.

The voltage divider on the ADC Pi reduces the 5V input to 2V so the ADC can read it accuratly which is why when you measured the output of the divider you saw 2V.

Have you checked your 5V source with a multimeter to see if it is exactly 5V? If the ADC Pi is measuring ~4.9V on all of the inputs it is possible that the 5V supply is actually ~4.9V instead of 5V. 

If the multimeter reading and the ADC reading are different you could compensate for this by calculating an offset value which you can apply to the ADC readings. For example if the voltage source is 5V and the reading is 4.9V you would divide 5 by 4.9 to give you an offset value of 1.02040816. You can then multiply the ADC reading by the offset to get the correct voltage.

`#!/usr/bin/env python3

 from ADCPi import ADCPi
 import time

 def main():

 adc = ADCPi(0x68, 0x69, 16)

    while (True):
        value=adc.read_voltage(1)
        value = value * 1.02040816
        print(value) # print the value
        time.sleep(1)  # Wait 1sec

 if __name__ == "__main__":
    main()
`

VK1IAN, 2024-12-13:
OK thanks. I'll try a different voltage source. I'm not at my bench so only have the +5V and GND on the ADC Pi to play with.
A few things.
> The MCP3424 is designed to support a maximum input voltage of 2.048 volts <
The datasheet I have from Microchip says:
>The on-board 2.048V reference voltage enables an input range of ± 2.048V differentially (full scale range = 4.096V/PGA).<
Given all the inputs on the board are wired as single ended, I assumed FSD would be 4.096V. The datasheet also specifies a maximum input range of VSS-0.3V to VDD+0.3V for all inputs so assumed it would be OK to use the 5V from the ADC Pi for testing.
With regard to your library, If 2V is applied to an analog input and the code prints 5V, I assume you have applied a "scaling factor" to the output??
I saw no reference to this in all your documentation or examples.

abelectronicsuk, 2024-12-13:
When the negative input on the MCP3424 is connected to ground the negative bit on the ADC readings is ignored so it effectively becomes a 17 bit ADC. Any voltages below 0V are not recorded so the voltage range becomes 0 - 2.048V.

You are correct that the ADC maximum voltage is VDD+0.3V so a 5V input should not damage the chip. 5V is very close to the limit so if the power supply for the Raspberry Pi is slightly below 5V or the voltage you are applying to the ADC input has any ripple it could exceed the threshold.

There is a scaling factor of 2.471 that is applied in the read_voltage function on line 271 in ADCPi.py. I did not include any mention in the documentation as the voltage range of the ADC Pi is advertised as 0 to 5V so the scaling factor is an integral part of the function for it go show the correct voltage.

VK1IAN, 2024-12-14:
> When the negative input on the MCP3424 is connected to ground the negative bit on the ADC readings is ignored so it effectively becomes a 17 bit ADC. Any voltages below 0V are not recorded so the voltage range becomes 0 - 2.048V.
> 
> You are correct that the ADC maximum voltage is VDD+0.3V so a 5V input should not damage the chip. 5V is very close to the limit so if the power supply for the Raspberry Pi is slightly below 5V or you used an external 5V source that had any ripple it could exceed the threshold.
> 
> There is a scaling factor of 2.471 that is applied in the read_voltage function on line 271 in ADCPi.py. I did not include any mention in the documentation as the voltage range of the ADC Pi is advertised as 0 to 5V so the scaling factor is an integral part of the function for it go show the correct voltage.

Thanks for information. Can I ask that this be included in your documentation please (i.e the 2.048V limit and the reason for the resistor divider)?? It would have saved me a lot of time yesterday.
I tried the input with a flat AAA battery and the ADC reading matched what I measured with a multimeter.

One more thing, the MCP3424 datasheet mentions that the config register is sent as the last byte in a "get" request. I don't see this mentioned anywhere in the ADCPi function list.
Accessing the config register on read is critical when using the device in one-shot mode, as the RDY bit indicates whether the conversion process has finished and has valid data.

abelectronicsuk, 2024-12-14:
I have updated the ADC Pi documentation with a note on the read_voltage function explaining the use of the voltage dividers and the scaling factor that is applied.

It is not possible to access the config register directly using our Python library as the register is used by several functions and changing it manually could break the functionality of the library.

You can change the conversion mode bit in the config register to one-shot by using the **set_conversion_mode(mode)** function. Passing a parameter of **0** will put the ADC into one-shot conversion mode, passing **1** will set it into continuous conversion mode.

VK1IAN, 2024-12-14:
I've been looking at the ADCPi.py library. Full disclosure; I am VERY new to python although I have used perl at work and C++ (arduino and similar). A lot of what I read didn't mean much, but I was able to understand most of it (more research required on my part).
Around line 271 (where you apply the scaling factor) I noticed that when in one-shot mode, the device was continually read and the config register checked until RDY indicates the conversion process has successfully completed. Only then is the result returned.
This was my only concern. I'll close this issue now.
Thank you for your time.

closed 2024-12-14 by VK1IAN